stm32-rs/stm32-eth #10: Trying to achieve max bandwidth results in (some) packets not being sent

birkenfeld, 2020-01-31:
Basically, we identified https://github.com/m-labs/smoltcp/issues/319 as not smoltcp's fault.

If you will have time to check this out with a reproducer, I will boil down my code to the bare minimum.

dtjones-atse, 2020-05-07:
I had an issue where when I compiled in release mode only every other ping was being responded to.
I ended up adding a Data Synchronization Barrier in the send method of Eth and that fixed it.
See the following change:
https://github.com/dtjones190/stm32-eth/commit/39d9d9b58f8dd65d959e777132ca6aada525b1fd#diff-b4aea3e418ccdb71239b96952d9cddb6

And the breadcrumbs that led me there:
https://community.st.com/s/question/0D50X00009Xkgff/stm32f746-ethernet-dma-transmit-problem
https://docs.rs/cortex-m/0.5.8/cortex_m/asm/fn.dsb.html

dtjones190, 2020-05-07:
I had an issue where when I compiled in release mode only every other ping was being responded to.
I ended up adding a Data Synchronization Barrier in the send method of Eth and that fixed it.
See the following change:
https://github.com/dtjones190/stm32-eth/commit/39d9d9b58f8dd65d959e777132ca6aada525b1fd#diff-b4aea3e418ccdb71239b96952d9cddb6

And the breadcrumbs that led me there:
https://community.st.com/s/question/0D50X00009Xkgff/stm32f746-ethernet-dma-transmit-problem
https://docs.rs/cortex-m/0.5.8/cortex_m/asm/fn.dsb.html

thalesfragoso, 2020-07-09:
@birkenfeld Are you using a F7 or F4 ? If using a F7, could you try the master branch to see if your problem persists ? This might also be caused by the lack of `compiler_fence`/`fence` in the proper places (even on F4s), I will hopefully address this on a PR soon.

birkenfeld, 2020-07-11:
I'm using an F429. Thanks for the suggestions, I will try master and stuff from #16 when I'm back at work.

thalesfragoso, 2020-07-13:
@birkenfeld You should probably test https://github.com/stm32-rs/stm32-eth/pull/17 instead of master now. Let me know if you find any problems.

birkenfeld, 2020-08-08:
Unfortunately, I tested with current master, with the "fence" feature both disabled and enabled, and could reproduce the problem.
I was using HCLK 180 MHz (see the PR I just submitted), but I see the same with 168 MHz.

Is #16 still relevant? Is there an example how to put the buffers in SRAM2?

thalesfragoso, 2020-08-09:
> Is #16 still relevant? Is there an example how to put the buffers in SRAM2?

Yes, it's still relevant, you can see an example here: https://github.com/rust-embedded/cortex-m-rt/pull/282, but you should use SRAM2/3 instead, also, beware that `cortex-m-rt` won't initialize these extra sections, so you need to do so, probably in [#[pre_init]](https://docs.rs/cortex-m-rt/0.6.12/cortex_m_rt/attr.pre_init.html). Not sure if that will fix the problem, but it can at least reduce contention, which seems important in your use case.

Could you provide the code (a reduced one, preferably) so we can see if we can reproduce it ? That together with your `memory.x`.

birkenfeld, 2020-08-09:
Will do; should be ready later today.

thalesfragoso, 2020-08-09:
Oh, I just realized that the `f4xx-hal` does enable the data cache, maybe you could try disabling it and see how it goes, however, for performance reasons, the best solution is indeed to leave the data cache enabled and use SRAM2/3.

birkenfeld, 2020-08-09:
I've put up the reproducing example as #21, I've not yet experimented with dcache.

thalesfragoso, 2020-08-10:
Ok, update to this issue:

Right now `smoltcp` puts the socket in a `Waiting` state after sending the ARP request, which causes the socket to stall indefinitely if we don't get a response, this isn't intended and a fixed is needed to make it re-send the ARP after some specific timeout.

Now, there is still one more question, do we ever send the ARP packet ? Because we tell `smoltcp` that we do, if we don't then there is a bug on our side (although it shouldn't alone cause the indefinitely stall). Another possibility is that the packet gets lost somewhere since the problem appears when dealing with a saturated link.

thalesfragoso, 2020-08-13:
Closing this as the problem was tracked down in smoltcp, issue here https://github.com/smoltcp-rs/smoltcp/issues/319, and fixes here https://github.com/smoltcp-rs/smoltcp/pull/368 and here https://github.com/smoltcp-rs/smoltcp/pull/369.

closed 2020-08-13 by thalesfragoso