stm32-rs/stm32-eth #16: Document possible problems when placing buffers on a cacheable region

thalesfragoso, 2020-07-09:
Quoting @adamgreig 
> if the dcache was enabled for the region the descriptor (or buffer) is in, it would need cleaning at this point (and the buffers would need invalidating on packet reception). Better to require that the descriptors and buffers are not placed in a dcache-enabled region, I think, and perhaps suggest they're placed in SRAM2 (even on F4) to reduce bus contention during DMA transfers.

astro, 2020-09-05:
Additionally, the buffers must not be placed in OCM. (That's not the same as a dcache-enabled region, right?)

adamgreig, 2020-09-05:
Do you mean 'CCM' (or 'DTCM' on STM32F7 and H7)? It's a bit more complicated: on STM32F4, you **can't** use the CCM ("core-coupled memory") for any DMA access; it's only accessible by the CPU so Ethernet (or any DMA) cannot read it. However on the F7 the DTCM _is_ exposed to the main memory bus _through_ the CPU, so the DMA engines **can** read it, but it's not a good use of the DTCM.

Both of those are totally distinct problems from not placing the buffers in a cacheable region, but typically both CCM and DTCM cannot be cached so at least you only suffer from one problem or the other.