stm32-rs/stm32-eth #32: Can't compile stm32-eth v0.2.0

NathanMcMillan54, 2021-06-17:
When I try to compile stm32-eth it can't compile the cortex-m dependencie. This error comes up three times and nothing else:

```command line
error[E0412]: cannot find type `RegisterBlock` in module `mpu`
   --> /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.2/src/peripheral/mod.rs:486:39
    |
486 |     pub const fn ptr() -> *const mpu::RegisterBlock {
    |                                       ^^^^^^^^^^^^^ not found in `mpu`
    |
help: consider importing one of these items
    |
58  | use crate::peripheral::cbp::RegisterBlock;
    |
58  | use crate::peripheral::cpuid::RegisterBlock;
    |
58  | use crate::peripheral::dcb::RegisterBlock;
    |
58  | use crate::peripheral::dwt::RegisterBlock;
    |
      and 7 other candidates
```

When I try to compile later version, the same error comes up. Is it my compiler? Or do the dependencies need to be updated?

adamgreig, 2021-06-17:
At a guess you're trying to build for your native target instead of a thumb embedded target, and additionally that target is not x86 (perhaps Apple Arm?). This would normally work (well, compile without error but not be functional) but there was a recent bug in cortex-m giving the error you're getting; it's been fixed in cortex-m but not yet released.

If you build with eg `--target thumbv7em-none-eabihf` does it work?

NathanMcMillan54, 2021-06-17:
@adamgreig, sorry I forgot to say I am using the ``thumbv7em-none-eabihf`` target.

adamgreig, 2021-06-18:
What command are you running to build, exactly, and what's in `.cargo/config` if any? The only way to get this error on cortex-m 0.7 is if your target is somehow not one of the thumb embedded ones or not x86, so that still seems the most likely candidate.

You can see in the 0.7.2 release commit here that the RegisterBlock is defined, but only for some architectures: https://github.com/rust-embedded/cortex-m/blob/653d218e39bb5dad14a0995f3753cbbd1569c9bf/src/peripheral/mpu.rs#L8

NathanMcMillan54, 2021-06-18:
This is my ``.cargo/config.toml``:

```toml
[target.'cfg(target_arch = "arm")']
rustflags = [
    "-C", "link-arg=--nmagic", "-C", "link-arg=-Tmemory.x"
]
```

And this is how I compile:

```command line
cargo build --target thumbv7em-none-eabihf
```

Should I added the build target to ```.cargo/config.toml```? (I'm on my phone right now so I can't try it)

adamgreig, 2021-06-18:
Huh, no, if you have `--target` on the commandline like that it should always take effect. You might be missing a `link-arg=-Tlink.x` though, unless your `memory.x` includes the full link script instead of just the memory sections.

If you clone this repository, can you build the examples? For instance, `cargo build --example="ip" --features="stm32f429 smoltcp-phy log smoltcp/socket-tcp smoltcp/socket-icmp smoltcp/log smoltcp/verbose"` (taken from the README). It works for me here but might help pin down what's going wrong for you. You could also check the RUSTFLAGS environment variable isn't set, just in case...

NathanMcMillan54, 2021-06-18:
I'll try that