stm32-rs/stm32-eth #44: Remove argument from stm32_eth::eth_interrupt_handler

thalesfragoso, 2022-06-04:
The function only does an "atomic" write, we could (unsafely) get the peripheral ourselves with `ptr()`.

This will allow us to remove some unsafe from our examples without adding more complexity to them. Besides, the unsafe in the example is only sound because we know the internals of the function, if the function's internals change, they could become unsound, we shouldn't be giving this type of example to our users.

datdenkikniet, 2022-06-12:
I think this change sounds like a good idea, but we should find a way to clearly document that the function is only race-condition free so long as no other code (within the library) writes to `eth_dma.dmasr`

datdenkikniet, 2023-02-17:
Fixed in #75

closed 2023-02-17 by datdenkikniet