stm32-rs/stm32-eth #79: stm32f429 (nucleo board) no longer working with examples on main branch

alios, 2023-03-12:
Hi!

I updated my code to the current main branch / smoltcp 0.9 and now ethernet is no longer working. Thought it might be a problem with my own code, but now tried the stm32-eth's examples and it is also no longer working. But if I switch back to the 0.4.x branch it starts working again. Primarily tried the `rtic-echo` example, but to exclude problems with rtic I also tried the `ip` example which also does not work. In both, rtic-echo and ip on the current master branch it's not only the sockets which seem not to work, the board even does not response to arp requests.

Here is the rtt output (with defmt's log level set to tracing) of the 0.4.x branch of the rtic-echo example:

```
DEFMT_LOG=trace PROBE_RUN_CHIP=STM32F429AGHx \                                                                                                               *[e6d2dea]
STM32_ETH_EXAMPLE_PINS=nucleo  \
cargo run --release --example rtic-echo \
    --features="stm32f4xx-hal stm32f429 defmt smoltcp-phy" \
    --target thumbv7em-none-eabihf
   Compiling defmt-macros v0.3.3
   Compiling defmt v0.3.2
   Compiling smoltcp v0.8.2
   Compiling panic-probe v0.3.0
   Compiling defmt-rtt v0.4.0
   Compiling stm32-eth v0.4.1 (/Users/alios/src/embedded-rust/stm32-eth)
    Finished release [optimized + debuginfo] target(s) in 3.52s
     Running `probe-run target/thumbv7em-none-eabihf/release/examples/rtic-echo`
(HOST) INFO  flashing program (63 pages / 63.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO  Pre-init
└─ rtic_echo::app::init @ examples/rtic-echo.rs:70
INFO  Setting up pins
└─ rtic_echo::app::init @ examples/rtic-echo.rs:80
INFO  Configuring ethernet
└─ rtic_echo::app::init @ examples/rtic-echo.rs:83
INFO  Enabling interrupts
└─ rtic_echo::app::init @ examples/rtic-echo.rs:94
INFO  Setting up smoltcp
└─ rtic_echo::app::init @ examples/rtic-echo.rs:97
TRACE [0]: adding
└─ smoltcp::iface::socket_set::{impl#2}::add::put @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:17
TRACE tcp:10.23.101.187:1337: state=Closed=>Listen
└─ smoltcp::socket::tcp::{impl#4}::set_state @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:17
INFO  Resetting PHY as an extra step. Type: LAN8742A
└─ rtic_echo::app::init @ examples/rtic-echo.rs:127
INFO  Waiting for link up.
└─ rtic_echo::app::init @ examples/rtic-echo.rs:134
INFO  Link up.
└─ rtic_echo::app::init @ examples/rtic-echo.rs:138
INFO  Detected link speed: FullDuplexBase100Tx
└─ rtic_echo::app::init @ examples/rtic-echo.rs:147
INFO  Setup done. Listening at (10.23.101.187, 1337)
└─ rtic_echo::app::init @ examples/rtic-echo.rs:155
DEBUG Got an ethernet interrupt! Reason: InterruptReasonSummary { is_rx: true, is_tx: false, is_error: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:172
DEBUG cannot process ingress packet: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Failed to consume RX token: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Got an ethernet interrupt! Reason: InterruptReasonSummary { is_rx: true, is_tx: false, is_error: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:172
DEBUG cannot process ingress packet: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Failed to consume RX token: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Got an ethernet interrupt! Reason: InterruptReasonSummary { is_rx: true, is_tx: false, is_error: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:172
DEBUG cannot process ingress packet: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Failed to consume RX token: Unrecognized
└─ smoltcp::iface::interface::{impl#2}::socket_ingress::{closure#0} @ /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.2/src/macros.rs:18
DEBUG Got an ethernet interrupt! Reason: InterruptReasonSummary { is_rx: true, is_tx: false, is_error: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:172
DEBUG Got an ethernet interrupt! Reason: InterruptReasonSummary { is_rx: true, is_tx: false, is_error: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:172
DEBUG cannot process ingress packet: Unrecognized
[... 220 lines cut ...]
INFO  Waiting for link up.
└─ rtic_echo::app::init @ examples/rtic-echo.rs:137
INFO  Link up.
└─ rtic_echo::app::init @ examples/rtic-echo.rs:141
INFO  Detected link speed: FullDuplexBase100Tx
└─ rtic_echo::app::init @ examples/rtic-echo.rs:150
INFO  Setup done. Listening at (10.0.0.1, 1337)
└─ rtic_echo::app::init @ examples/rtic-echo.rs:158
DEBUG Got an ethernet interrupt! Reason: InterruptReason { rx: true, tx: false, dma_error: false, time_passed: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:183
DEBUG Got an ethernet interrupt! Reason: InterruptReason { rx: true, tx: false, dma_error: false, time_passed: false }
└─ rtic_echo::app::eth_interrupt @ examples/rtic-echo.rs:183
^C────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: cortex_m::asm::nop
        at /Users/alios/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.7/src/call_asm.rs:19:21
   1: main
        at examples/rtic-echo.rs:21:1
   2: Reset
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) INFO  device halted by user
```

Did anyone else experience that with a stm32-f4 device? 
I'am not deep enough into this project or smoltcp to debug this any further.

roel0, 2023-03-17:
I can reproduce on stm32f437. 0.4.1 works fine, HEAD of master I don;t receive ANY interrupts

datdenkikniet, 2023-03-22:
Hmm, that's... Concerning. Could you guys try it with 9987cbeec90e62aeda0329e7547f29ac3244a074? I think it may be the async-await support, it's been working for me this far on an F7 but I haven't been able to test it on an F4.

alios, 2023-03-23:
@datdenkikniet with 9987cbe everything works just fine - so I guess your idea might be correct.

datdenkikniet, 2023-03-23:
Alright, I have absolutely now idea how it got through, or how it ever worked for me, but I may have found the problem.

Please try with https://github.com/stm32-rs/stm32-eth/pull/80 and report.

alios, 2023-03-23:
@datdenkikniet Tested the rtic-echo example again as well as my own code base and with the new master and the patch from #80 everything seems to work again. 

Big thnx

closed 2023-03-23 by alios

datdenkikniet, 2023-03-23:
Awesome! Thanks again for reporting and confirming to both you and @roel0